dynawo/dynawo #2613: Dynawo distrib header does not forward python path

jlbouchot, 2022-11-30:
Assuming python is defined through another alias than just python, it is required to set the environment variable DYNAWO_PYTHON_COMMAND in the myEnvDynawo.sh file 

This is not forwarded to the distributionFolder/dynawo.sh file under the variable export_var_env DYNAWO_PYTHON_COMMAND=python

rosiereflo, 2022-12-02:
Normally it should be done a the line 71 of dynawo.sh

`  export_var_env DYNAWO_PYTHON_COMMAND=python`

rosiereflo, 2022-12-06:
Closing the issue, please feel free to reopen it if I didn't answer your question

jlbouchot, 2022-12-24:
I agree _an_ import is done in the dynawo.sh. But it does not take into account the environment algorithm defined in the myEnvDynawo.sh used in the command line. i.e. Independently whether or not the variable is set, such as `export DYNAWO_PYTHON_COMMAND="python3"`, the variable at line 71 in the dynawo.sh will read 
`export_var_env DYNAWO_PYTHON_COMMAND=python`.

rosiereflo, 2023-01-23:
I wasn't able to reproduce with the distrib:

> ./dynawo.sh jobs-with-curves dynawo/nrt/data/IEEE14/IEEE14_SyntaxExamples/IEEE14_BlackBoxModels/IEEE14.jobs 
[INFO] xsd validation will not be used
launching job 'IEEE14 - Black box models'
job 'IEEE14 - Black box models' succeeded
results are available in dynawo/nrt/data/IEEE14/IEEE14_SyntaxExamples/IEEE14_BlackBoxModels/outputs
Generating curves visualization pages
Generating curves visualization page for job <IEEE14 - Black box models> ...
... End of generating curves visualization page for job <IEEE14 - Black box models>
End of generating curves visualization pages

> export DYNAWO_PYTHON_COMMAND="bla"

> ./dynawo.sh jobs-with-curves dynawo/nrt/data/IEEE14/IEEE14_SyntaxExamples/IEEE14_BlackBoxModels/IEEE14.jobs 
[INFO] xsd validation will not be used
launching job 'IEEE14 - Black box models'
job 'IEEE14 - Black box models' succeeded
results are available in dynawo/nrt/data/IEEE14/IEEE14_SyntaxExamples/IEEE14_BlackBoxModels/outputs
Generating curves visualization pages
./dynawo.sh: ligne 94: bla : commande introuvable
./dynawo.sh: ligne 94: bla : commande introuvable
Error during curves visualisation page generation

SO looks like the environment variable has the correct value during execution. What was the error that you had?

closed 2023-01-25 by rosiereflo