dynawo/dynawo #2619: Fix dynawo

cossartque, 2022-12-01:


cossartque, 2022-12-01:
The compilation works but the results are not as expected.
@GerardThevenon can you give the test case to @rosiereflo ?
Thanks

GerardThevenon, 2022-12-07:
Build failed on [2619_dynamic_indexing](https://github.com/dynawo/dynawo/tree/2619_dynamic_indexing)

(IECWT4ACurrentSource)

rosiereflo, 2022-12-07:
> Build failed on [2619_dynamic_indexing](https://github.com/dynawo/dynawo/tree/2619_dynamic_indexing)
> 
> (IECWT4ACurrentSource which uses Multiswitch)

Yup, a bug that I introduced at the last moment. Let me fix that

rosiereflo, 2022-12-08:
should be ok now

GerardThevenon, 2022-12-08:
> should be ok now

Unfortunately it looks as if the index doesn't take into account the operation "+1" defined in MultiSwitch.mo
Also the behavior of a step is questionable. Here are some results :

[image]
[image]
[image]
[image]
[image]

rosiereflo, 2022-12-09:
Ready for a new test!

GerardThevenon, 2022-12-09:
I got the same result.

rosiereflo, 2022-12-09:
Are you sure the models were recompiled ?

GerardThevenon, 2022-12-09:
I didn't want to build-all so I compiled dynawo-build-models IECMultiSwitch

GerardThevenon, 2022-12-12:
In the end, it works as intended after clean-build-all !!

commit 8b3c2bc8 (2022-12-09): #2619 fix multi switch
closes #2619
commit 62d3aac0 (2022-12-12): #2619 fix multi switch
closes #2619
commit d1f1a61e (2022-12-14): #2619 fix multi switch
closes #2619
commit c56f06a6 (2022-12-14): #2619 fix multi switch
closes #2619