dynawo/dynawo #2765: Derivative MSL block vs der operator

ErwanGuichard49, 2023-01-31:
On GitLab branch **521_add_inelfe**, the model _dynawo_rte/dynawo/sources/Models/Modelica/DynawoRTE/Electrical/DC/Specific/Inelfe/BaseControls/DcSiem.mo_ prevents the simulation _dynawo-rte/nrt/data/HVDC/Inelfe/HVDC.jobs_ from being run.

Using the operator **der** yields at the beginning of the simulation the following error

the number of algebraic/differential variables is different from the number of algebraic/differential equations in the simulated problem

If the equations using **der** are replaced by the MSL Derivative block, the simulation runs smoothly, even if the operator **der** is used in other equations.

To reproduce the issue, uncomment lines 64 and 65, comment lines 51, 52 and 66 to 69.

rosiereflo, 2023-02-09:
The error message is correct: with the original equations, this models contains 3 derivative variables for 5 differential equations whereas Dynawo makes the assumption that the system is squared for algebraic and differential variables/equations.
using the Modelica.Blocks.Continuous.Derivative simply adds 2 differential equations, solving this issue. Another possibility is to use 2 intermediate algebraic variables Real der = der(var1).

ErwanGuichard49, 2023-02-09:
Addng two algebraic variables derIDci = der(IDci) (i = 1 or 2) enables the simulation to run smoothly.

closed 2023-02-09 by ErwanGuichard49