dynawo/dynawo #2771: Calculation error by floor function

ErwanGuichard49, 2023-02-02:
Follow-up to issue https://github.com/dynawo/dynawo/issues/2696 : on GitLab branch 570, in the dynawo-rte/dynawo/sources/Models/Modelica/DynawoRTE/Electrical/DC/Specific/Inelfe/BaseControls_INIT/TfTap_INIT.mo model, the input of the function floor is the output of a CombiTable1Ds.

The input is well calculated by the block CombiTable1Ds and increased by 0.5 (in HVDCInelfe.mo, b10 = 11.105, b20 = 9.94081), but the result of the floor function is 0 on both sides, instead of the expected values b10 = 11 and b20 = 9.

rosiereflo, 2023-02-03:
The issue here is that the variables with the floor value (e.g. Real v = floor(Real other)) are  continuous, so there is no root raised (and no discrete variable updated) when the floor value changes, leading to this error. @adriguir any opinion on this?

ErwanGuichard49, 2023-02-03:
The issue can be solved by writing in a when structure  the equation between the continuous variable and the discrete variable. This new structure can be seen on branch 570, which has been updated.

ErwanGuichard49, 2023-02-06:
Another way of implementing this is to use a CombiTable1Ds with the "ConstantSegments" interpolation method.

closed 2023-02-06 by ErwanGuichard49