dynawo/dynawo #2787: [Models compilation] Allow to use operator noEvent on methods not defined on R  (e.g. sqrt)

rosiereflo, 2023-02-09:
should solve #2611

adriguir, 2023-02-10:
It would be really nice to have it quite soon ;) Have you a first idea of the necessary effort?

rosiereflo, 2023-02-10:
on going! should take a few days

FredericSabot, 2023-07-19:
Hi, the wind current limits should be updated too. The following seems to work for me.

```
iqMaxPu = min(iqVdlPu, noEvent(if(abs(IMaxPu) > abs(min(ipLimPu, ipVdlPu))) then sqrt(IMaxPu ^ 2 - min(ipLimPu, ipVdlPu) ^ 2) else 0));
    iqMinPu = -iqMaxPu;
  else
    if time <= vDipFrzEndTime and vDipFrzEndTime >= 0 then
      ipMaxPu = ipMaxFrzPu;
    else
      ipMaxPu = min(ipVdlPu, noEvent(if(abs(IMaxPu) > abs(min(iqLimPu, iqVdlPu))) then sqrt(IMaxPu ^ 2 - min(iqLimPu, iqVdlPu) ^ 2) else 0));
    end if;
```

FredericSabot, 2023-07-19:
Spoke too fast, it works for some cases but not all.

FredericSabot, 2023-08-11:
With the current implementation, I sometimes still get the ```103_PV_1 : Model error: Argument of sqrt(ibg.limitUpdating.IMaxPu ^ 2.0 - ibg.limitUpdating.IpLimPu ^ 2.0) was %g should be >= 0 -0.376634``` error (using GenericIBG that has basically the same limit updating loop as the WECC photovoltaics) when using the simplified solver.

The error disappears if I use a very small timestep (hmax = 1e-4) or if I use IDA. However, it should normally work no matter the timestep. Do you also have such cases? I can provide a test case if useful.

rosiereflo, 2023-09-29:
I updated the branch in the PR, you case seem to pass now

FredericSabot, 2023-10-02:
Hi, I will test it and see if I can generate a test case that doesn't pass early next week then.

closed 2023-10-03 by marcochiaramello

FredericSabot, 2023-10-12:
It works well for a large set of cases with both IDA and solverSIM!

adriguir, 2023-10-12:
Great! I'm also quite impatient to play with it, but that should really open a lot more possibilities for systems with a high share of PE-based components :) I hope I will have a bit of time to do some !

commit a125a79c (2023-09-29): #2787 handle noEvent
closes #2787
PR #2795 (2023-10-03): #2787 handle noEvent
closes #2787

**Checklist before requesting a review**

*use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes*

- [ ] unit tests and non-regression tests were added (new model, new feature and bug fix)
- [ ] main documentation was updated (update of input/output file, 3rd party, model, repository organization, solver)
- [ ] example documentations were updated (new example in examples folder)
- [ ] if this PR modifies the parameters or inputs/outputs of a model/solver: the corresponding xsl was added in util/xsl and platform DB were updated
-  [... cut]
commit e8bb9bc3 (2023-10-03): #2787 handle noEvent
closes #2787