dynawo/dynawo #2797: [Modelica compiler] Improve support for boolean vectors

FredericSabot, 2023-02-17:
Support of the functions in Modelica.Math.BooleanVectors would allow cleaner implementation of the model in PR #2573 

In particular, anyTrue() is needed for distance protection and LVRT, and index() is needed for UFLS.

Note that index requires other functions as shown in the logs in https://github.com/dynawo/dynawo/pull/2573#discussion_r1109724367

rosiereflo, 2023-02-20:
Hi, the method anyTrue is supposed to be supported since #2263

FredericSabot, 2023-02-20:
Indeed, actually the issue (regarding anyTrue) is its usage in a when condition:

`when anyTrue(tripped) then`

does not compile, but

```
trippedOR = anyTrue(tripped)
when trippedOR then
```

does.

rosiereflo, 2023-02-20:
ok thanks let me take a look

FredericSabot, 2023-04-18:
Ideally, support should also be added for
```
deltaPQ = -sum(UFLSStep[index(stepActivated)]);
```

rosiereflo, 2023-04-21:
> Ideally, support should also be added for
> 
> ```
> deltaPQ = -sum(UFLSStep[index(stepActivated)]);
> ```

Hi, would it be possible for you to provide me a model demonstrating the issue? Thanks!

commit b524bee5 (2023-03-15): #2797 improve handle of anyTrue when used in evalG
closes #2797
PR #2824 (2023-03-23): #2797 improve handle of anyTrue when used in evalG
closes #2797

**Checklist before requesting a review**

*use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes*

- [ ] unit tests and non-regression tests were added (new model, new feature and bug fix)
- [ ] main documentation was updated (update of input/output file, 3rd party, model, repository organization, solver)
- [ ] example documentations were updated (new example in examples folder)
- [ ] if this PR modifies the parameters or inputs/outputs of a model/solver: the corresponding xsl was added in util/xsl a [... cut]
commit 731ea57b (2023-03-23): #2797 improve handle of anyTrue when used in evalG
closes #2797
commit 945c6f52 (2023-07-07): #2797 better support for boolean vectors
closes #2797
closed 2023-07-24 by rosiereflo