dynawo/dynawo #2888: KINSOL fails to solve after equation type change following a disconnection

FredericSabot, 2023-04-24:
KINSOL is unable to solve the system (with WARN | ilegal input in the logs) if an equation like

```
der(omega) * H = Ce - Cm
```
becomes
```
omega = 0
```
e.g. after a switchoff.

This can be worked around as shown in https://github.com/FredericSabot/dynawo/commit/6ed531aff9ea9352bdae72e939a8e9b8d1549de0, but it is not very intuitive and it is difficult to identify when/where such issue occurs.

rosiereflo, 2023-04-24:
Hi, is this with the simplified solver? Would you have a testcase?

FredericSabot, 2023-04-24:
Both with the simplified solver and IDA, here is a test case which only requires code from the main branch (hopefully)

[IEEE39_KINSOL_issue.zip](https://github.com/dynawo/dynawo/files/11309779/IEEE39_KINSOL_issue.zip)

rosiereflo, 2023-06-02:
Hi, this is an issue in OpenModelica frontend that shows up during the LoadAlphaBetaMotor compilation. 
Depending on the order of the equations in the two branches of   `if  (running.value) ... else ... end if;` the code generated may or may not contains a hardcoded division by zero. 

For example I was able to simulate your example with Dynawo by simply moving the equation `omegaRPu = 0;` just before `terminal.i = Complex(0);`.
I was also able to reproduce the issue ijn a pure OpenModelica environment.

I will create a ticket on their side.

FredericSabot, 2023-06-02:
Great, thanks!

rosiereflo, 2023-06-02:
Ticket https://github.com/OpenModelica/OpenModelica/issues/10787

closed 2023-06-02 by rosiereflo