dynawo/dynawo #2898: [SolverSIM/TRAP] Solver does not detect ramps

FredericSabot, 2023-04-26:
Maybe this is expected, but solverSIM does not detect ramps, i.e.

[image]

becomes this

[image]

This can be solved by setting optimizeAlgebraicResidualsEvaluations to false.

Attached is a test case with code from https://github.com/FredericSabot/dynawo/tree/22_ISGT2023

[reduced_95_offpeak_frequencytest.zip](https://github.com/dynawo/dynawo/files/11330493/reduced_95_offpeak_frequencytest.zip)

rosiereflo, 2023-06-05:
Hi, it seems that the model AggregatedIBG is missing in the provided branch whereas it is used in the testcase, could you please provide a pointer to it? Thanks!

FredericSabot, 2023-06-05:
Hello, the model AggregatedIBG does well exist in the 22_ISGT2023 branch (https://github.com/FredericSabot/dynawo/blob/22_ISGT2023/dynawo/sources/Models/Modelica/PreassembledModels/AggregatedIBG.xml)

rosiereflo, 2023-06-21:
Hi, this is a bug due to the fact that this optimization was missing the specific case of algebraic equations depending on time. I will fix this.

PR #2964 (2023-07-03): #2898 reevaluate algebraic F after an algebraic restoration to take into
account algebraic equyations depending on time
closes #2898

**Checklist before requesting a review**

*use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes*

- [ ] unit tests and non-regression tests were added (new model, new feature and bug fix)
- [ ] main documentation was updated (update of input/output file, 3rd party, model, repository organization, solver)
- [ ] example documentations were updated (new example in examples folder)
- [ ] the corresponding milestone was added in the tic [... cut]
commit f1fbfa1e (2023-07-03): #2898 reevaluate algebraic F after an algebraic restoration to take into
account algebraic equyations depending on time
closes #2898