dynawo/dynawo #2909: Investigate why the dynawo precision had to be put to 1e-5 in DynaFlow

cossartque, 2023-05-02:
Since the refactoring of the generator models + addition of pStatus / Timeline (OS 2821), the default precision of 1e-4 doesn't work anymore for all the recollements.

The refactoring in itself doesn't change anything.
The main difference is the addition of a pStatus variable + when for the timelines.

cossartque, 2023-05-11:
This is linked to the calculation of limUQUp and limUQDown as a function of qStatus, direclty in the base model. (limUQUp = qStatus == GenerationMax for example). To be investigated in the future if we want to do it that way. Otherwise, the implementation where limUQUp and Down do not depend on qStatus works fine.

rosiereflo, 2023-05-11:
The issue is linked to the fact that Jacobian is recomputed when a boolean is modified when it is not when an enum (such as qstatus). Investigation on why are ongoing.

closed 2023-05-12 by rosiereflo