dynawo/dynawo #2945: SwitchOffSignals/Running Handling - Initially disconnected elements

adriguir, 2023-06-06:
In the current implementation of the SwitchOff control blocks, an element cannot be disconnected at the beginning of the simulation.
Indeed, running is initialized at true and thus even if someone writes something like:
shuntB.switchOffSignal1.value = false;
the shunt will remain connected all over the simulation.

I have seen some developments on this topic so maybe it no longer applies. Otherwise it would be nice to have running0 as a parameter (State0 doesn't have any impact on the running status...)

@joyelfeghali @cossartque FYI

cossartque, 2023-06-14:
I'm not sure I understand the question but if you write shuntB.switchOffSignal1.value = true; (and not false), it will be disconnected.

adriguir, 2023-06-15:
There was a typo indeed: the point is that even if I write this shuntB.switchOffSignal1.value = true;, it remains connected.
For example, if I take the SMIBStepPm example from the Modelica library,  and I modify the switchOffSignal for one transformer in the following manner: 

  transformer.switchOffSignal1.value = true;
  transformer.switchOffSignal2.value = false;

When I look to the plots, I get the following P,Q flows through the transformer (not zero):

[image]

In my understanding, there is no way in the initialization scheme to have running equal to false at the t = 0. And then the when equations driving any update of running are inhibited because switchOffValue1 and switchOffValue2 or 3 are initially true so there is no change during the simulation.

  if (NbSwitchOffSignals >= 3) then
    when switchOffSignal1.value or switchOffSignal2.value or switchOffSignal3.value and pre(running.value) then
      running.value = false;
    elsewhen not switchOffSignal1.value and not switchOffSignal2.value and not switchOffSignal3.value and not pre(running.value) then
      running.value = true;
    end when;
  elseif (NbSwitchOffSignals >= 2) then
    when switchOffSignal1.value or switchOffSignal2.value and pre(running.value) then
      running.value = false;
    elsewhen not switchOffSignal1.value and not switchOffSignal2.value and not pre(running.value) then
      running.value = true;
    end when;
  else
    when switchOffSignal1.value and pre(running.value) then
      running.value = false;
    elsewhen not switchOffSignal1.value and not pre(running.value) then
      running.value = true;
    end when;
  end if;

With Thibault, we had to do something like this to make it work:

when time < 1 then
  transformer.switchOffSignal1.value = false;
else when time >=1 then
   transformer.switchOffSignal1.value = true;
end when;

We haven't found any way to begin with a disconnected component, but maybe there is one!

adriguir, 2023-06-15:
NB: I can also put the two switchOffSignals equal to true, the transformer is still connected.

ErwanGuichard49, 2023-09-15:
Please check branch _2945_switchOff_init_. Does it solve this issue ?

PR #3039 (2023-11-07): #2945 Adding initial parameters for switchOffSignals
closes #2945
closed 2023-11-07 by joyelfeghali