dynawo/dynawo #2974: NodeFault with a nodeBreaker topology not handled well

gautierbureau, 2023-06-30:
``` bash
<dyn:blackBoxModel id="FAULT" lib="NodeFault" parFile="file.par" parId="fault"/>
<dyn:macroConnector id="FaultToNode" name2="true">
  <dyn:connect var1="fault_terminal" var2="@NAME@_ACPIN"/>
</dyn:macroConnector>
<dyn:macroConnect connector="FaultToNode" id1="FAULT" id2="NETWORK" name2="BBS"/>
```

The issue is the name2 we have to provide, if we give the name of a BusBarSection we need to add ```<par type="BOOL" name="BBS_hasShortCircuitCapabilities" value="true"/>``` but the BBS id is transformed into dynawo with `calculatedBus_` prefix. And even if we have a load or generator connected to the BBS, that should activate the `hasConnection_` of the following `if (hasConnection_ || hasShortCircuitCapabilities_)` in the `DYNModelBus.cpp` it is not the case because once again we have a calculatedBus for the generator and for the BSS so they are not the same. We need to test properly on a BBS that has only lines and a BSS that has loads or generators to test the two conditions of the if. 
We can also think of a global parameter to expose the i variables globally when we need it.

rosiereflo, 2023-07-03:
This should be taken care of with the @NODE@ macro of the macroConnect. Does it work when using it?

rosiereflo, 2023-07-03:
Regarding the hasShortCircuitCapabilities_ the idea it that it is up to the platform to generate the proper parameter to activate it

gautierbureau, 2023-07-03:
> This should be taken care of with the @node@ macro of the macroConnect. Does it work when using it?

`"@@NAME@@@NODE@_ACPIN"` could work but again we need to give the name of the calculatedBus and not the BBS has dynawo cannot retrieve a bus with a BBS. 

For the parameter it is the same issue `BBS_1_hasShortCircuitCapabilities` would not work but `calculatedBus_BBS_0_hasShortCircuitCapabilities` would...

gautierbureau, 2023-07-03:
In `getBusName` we have the names of CalculatedBus not BBS so the replace inside `findNodeConnectorName` cannot work

rosiereflo, 2023-07-03:
We are modeling buses not bbs so I don't understand the last comment. However it is true that we need a mechanism to link a bbs to a calculated bus for the paramter

gautierbureau, 2023-07-03:
> We are modeling buses not bbs so I don't understand the last comment. However it is true that we need a mechanism to link a bbs to a calculated bus for the paramter

I know but to write the dyd you only know BBS and not the Bus created later on hence the issue...

closed 2023-07-10 by rosiereflo