dynawo/dynawo #2998: bus_hasShortCircuitCapabilities use

gautierbureau, 2023-07-10:
For me it seemed it was supposed to be used to activate the capability on all buses, but if I do this I end up with the following error `DYN Error: variables number 613 not equals to the equations number 515`.

rosiereflo, 2023-07-10:
if some buses do not have this capability the <STATIC_ID>_hasShortCircuitCapabilities parameter should be used.

gautierbureau, 2023-07-10:
> if some buses do not have this capability the <STATIC_ID>_hasShortCircuitCapabilities parameter should be used.

Yes so my point is this parameter is useless...

rosiereflo, 2023-07-10:
It was created for the small example cases and also to be homogeneous: all the network parameters have a global and local versions

gautierbureau, 2023-07-10:
But in this case it is not usable in a global way so I don't see why allow it...

rosiereflo, 2023-07-10:
In the end it does activate the capability for all buses but it means that the testcase should contain additional connexions on all buses. I don't know if it's a need that exists or not

FredericSabot, 2023-07-13:
It would be nice to be able to activate the hasShortCircuitCapabilities for all buses without having to add connections to all of them. The need is in systematic analyses (e.g. single fault but at different buses in each scenario) and general convenience.

The current alternative of adding dummy loads at all buses / in the event.dyd is fine but may seen like black magic to a new user. (On this note, there seems to be no mention of hasShortCircuitCapabilities in the main doc. I would even argue that it would be best if the user did not have to know about this.)