dynawo/dynawo #3091: Add IEEE regulations for generators outside France

ErwanGuichard49, 2023-12-01:


joyelfeghali, 2024-02-14:
Developments of this issue will cover the following models:
Governors:
- HyGov (updated version)
- Gast
- IEEEG1 (updated version)
- IEEEG2
- TGov1 (updated version)
- TGov3

IEEE PSS:
- PSS1A
- PSS2A (updated version)
- PSS2C
- PSS3B
- PSS3C
- PSS6C

Voltage regulators:
- AC1A
- AC1C
- AC6A
- AC6C
- AC7B
- AC7C
- AC8B
- AC8C
- DC1A
- ST1A
- ST1C
- ST4B (updated version)
- ST4C
- ST6B
- ST6C
- ST7B
- ST7C
- ST9C
- BBSEX1
- IEEEX2
- IEEEX2A
- SCRX (updated version)

martinscripts, 2024-05-22:
Hey @ErwanGuichard49, I am having a problem with the DC1A model. 
It extends DC1C, and actually there is the problem.

In DC1C, the line 63:
```
  Modelica.Blocks.Continuous.TransferFunction transferFunction(a = {tB, 1}, b = {tC, 1}, x_scaled(start = {Va0Pu / Ka}), x_start = {Va0Pu / Ka}, y(start = Va0Pu / Ka)) annotation(
```

x_scaled is a protected variable in TransferFunction.mo.
OMEdit deals with this somehow, but when using OMPython, the model can't compile. It throws an arror, because in `x_scaled(start = {Va0Pu / Ka})` a protected variable is attempted to be modified. 

I changed line 63 to 
```
  Modelica.Blocks.Continuous.TransferFunction transferFunction(a = {tB, 1}, b = {tC, 1}, x_start = {Va0Pu / Ka}, y(start = Va0Pu / Ka)) annotation(
```

And it works. 
Do you think this has other side effects? (start value is given in `y(start=...)` and `x_start=...`)

If this is fine, I would suggest to change it.

ErwanGuichard49, 2024-05-23:
> Hey @ErwanGuichard49, I am having a problem with the DC1A model. It extends DC1C, and actually there is the problem.
> 
> In DC1C, the line 63:
> 
> ```
>   Modelica.Blocks.Continuous.TransferFunction transferFunction(a = {tB, 1}, b = {tC, 1}, x_scaled(start = {Va0Pu / Ka}), x_start = {Va0Pu / Ka}, y(start = Va0Pu / Ka)) annotation(
> ```
> 
> x_scaled is a protected variable in TransferFunction.mo. OMEdit deals with this somehow, but when using OMPython, the model can't compile. It throws an error, because in `x_scaled(start = {Va0Pu / Ka})` a protected variable is attempted to be modified.
> 
> I changed line 63 to
> 
> ```
>   Modelica.Blocks.Continuous.TransferFunction transferFunction(a = {tB, 1}, b = {tC, 1}, x_start = {Va0Pu / Ka}, y(start = Va0Pu / Ka)) annotation(
> ```
> 
> And it works. Do you think this has other side effects? (start value is given in `y(start=...)` and `x_start=...`)
> 
> If this is fine, I would suggest to change it.

The TransferFunction block from the Modelica Standard Library does not give a start value to x_scaled. Under OpenModelica, if initType = InitialState, x_scaled is correctly initialized with x_start. Under Dynawo, the initial equations are ignored, making it necessary to set a start value for x_scaled. Therefore, your proposal cannot be integrated.

Nevertheless, I am currently working on a new TransferFunction block with a cleaner initialization based on start values. It would solve the issue with OMPython.

martinscripts, 2024-05-23:
hey @ErwanGuichard49 , thanks for the quick answer. 
Great, I am looking forward to the new block, because this issue makes the library less usable for me since I test my developments using OMPython. 
If I can somehow support with that, let me know. For now, I will use a duplicate version of dc1a to make it work for me.

closed 2024-05-24 by joyelfeghali

ErwanGuichard49, 2024-05-27:
> hey @ErwanGuichard49 , thanks for the quick answer. Great, I am looking forward to the new block, because this issue makes the library less usable for me since I test my developments using OMPython. [...]

The new block Dynawo.NonElectrical.Blocks.Continuous.TransferFunction has been added to the library last Friday. It is used for the moment in Pss3c only. In order to solve the issue with OMPython, you can write the line 63 of Dc1c as follows :

Dynawo.NonElectrical.Blocks.Continuous.TransferFunction transferFunction(a = {tB, 1}, b = {tC, 1}, x_start = {Va0Pu / Ka}, y_start = Va0Pu / Ka);