dynawo/dynawo #3095: Parameter not propagated in Pss2c

ErwanGuichard49, 2023-12-11:
On dynawo branch **3091_add_ieee_regulations_foreign_generators**, in the model dynawo/sources/Models/Modelica/Dynawo/Electrical/Controls/Machines/PowerSystemStabilizers/Standard/Pss2c.mo, the parameter M has a default value equal to 5 (line 22).

This parameter ought to be propagated :
- first to rampTrackingFilter (line 7)
- then to leadMOrderLagCascade (line 28 of dynawo/sources/Models/Modelica/Dynawo/NonElectrical/Blocks/Continuous/RampTrackingFilter.mo)
- where it defines the number of firstOrderCascade blocks (line 25 of dynawo/sources/Models/Modelica/Dynawo/NonElectrical/Blocks/Continuous/LeadMOrderLag.mo) and their connections (in the for loop on lines 45 to 47)

On dynawo-rte branch **734_add_swiss_regulations**, in the test case dynawo-rte/nrt/data/SMIB/SMIB_Fault_LessThan_tCritical/SMIB_GoverChVRUnitrolCh/SMIB.jobs, the Pss2c model is used and the parameter M is modified (in SMIB.par) to M = 6. This value is taken into account in Pss2c but is not propagated to the submodels RampTrackingFilter and LeadMOrderLag, where it remains at the default value of 5, as seen when plotting M at these different levels.

rosiereflo, 2023-12-14:
Hi, the default value is propagated to all parameters, however the parameters remain distinct and with the current implementation the user needs to provide the new value of M for each submodel using it. Maybe you could try to remove the default value in the submodels?

ErwanGuichard49, 2023-12-15:
With the default values removed from the submodels, the highest default value (M = 5) is propagated to the submodels. Adding new parameter lines in SMIB.par changes the value of M as plotted (M = 6) but leaves the simulation untouched, meaning that M is still assumed to be equal to 5 by the compiler.

ErwanGuichard49, 2023-12-18:
I have pushed a provisional fix to the problem on branch **3091_add_ieee_regulations_foreign_generators**. The models RampTrackingFilter.mo and LeadMOrderLag.mo have been modified :

- Instead of declaring a vector of blocks of undetermined size (_firstOrderCascade[max(0, M-1)]_), a vector of fixed size is now declared (_firstOrderCascade[5]_)

- In the if structures, the connect statements have been replaced by equations

- Inconveniently, it is impossible to compile successfully under dynawo-rte an equation (in an if structure) y = firstOrderCascade[i].y. I have thus detailed the equation for every value of M (see LeadMOrderLag.mo, lines 46 to 55).

ErwanGuichard49, 2023-12-18:
The code ought to be compiled effectively with the lines 44 and 45 of LeadMOrderLag.mo uncommented and the lines 46 to 55 commented.

ErwanGuichard49, 2023-12-20:
In the newest implementation of the models and of the case, the value of M (parameter of LeadMOrderLag.mo) is set to 6 but this value is not taken into account in the simulation. Instead, the default value set in the highest model (Pss2c) is used : 5. The following if structure :

  if M == 1 then
    connect(leadlag.y, y);
  else
    connect(firstOrderCascade[M-1].y, y);
  end if;
  
ought to yield connect(firstOrderCascade[5].y, y) but actually (as seen on the curves) becomes connect(firstOrderCascade[4].y, y)

Intriguingly, if plotted, M appears as 6.

rosiereflo, 2024-01-09:
Hi, the table indexes of Dynawo are from 0 to N-1 when Modelica are from 1 to N. firstOrderCascade[4].y from Dynawo is thus firstOrderCascade[5].y of Modelica, and thus the connection seems to be the one expected

ErwanGuichard49, 2024-01-09:
Whatever value is given to M in dynawo-rte/nrt/data/SMIB/SMIB_Fault_LessThan_tCritical/SMIB_GoverChVRUnitrolCh/SMIB.par (lines 122 to 124), the output y of the rampTrackingFilter block is equal to firstOrderCascade__3_y (Dynawo index) i.e. firstOrderCascade[4].y (Modelica index). This result is consistent with M = 5, the default value given in Pss2c.mo.

If M = -1, the assert statement on line 33 of LeadMOrderLag.mo is not triggered : the value of M from the .par is not taken into account in the simulation but it is displayed correctly on a graph.

ErwanGuichard49, 2024-01-10:
The issue has been solved, albeit inelegantly, by creating an intermediate variable vector named z and filled b a for loop :

  for m in 2:MMax loop
    if m == M then
      z[m-1] = firstOrderCascade[m-1].y;
    else
      z[m-1] = 0;
    end if;
  end for;

  if M == 1 then
    y = leadlag.y;
  else
    y = sum(z);
  end if;
  
Thus, if M<>1, y = firstOrderCascade[M-1].y. Writing this directly causes the dynawoRTE compilation to fail : the M-1 th component did not exist yet.

The new design has been pushed on branch **3091_add_ieee_regulations_foreign_generators**.

closed 2024-01-15 by ErwanGuichard49