dynawo/dynawo #3097: Argument of sqrt was negative

GerardThevenon, 2023-12-13:
On branch #2997 I have created a new preassembled _WTG4AWeccVoltageSource_ and a test case _nrt/data/WECCAdditionalModels/Wind/WTG4AVoltageSource/WTG4A.jobs_

The error happens at the end of a fault : _Model error: Argument of sqrt(WTG4A.reec.currentLimitsCalculation.IMaxPu ^ 2.0 - WTG4A.reec.iqCmdPu ^ 2.0) was %g should be >= 0 -2.22105_

Please note that the block called _CurrentLimitsCalculationA_ is already used in other test cases but the simulations are OK there.

rosiereflo, 2023-12-14:
COuld you please show the Modelica block where this sqrt is used?

GerardThevenon, 2023-12-14:
> COuld you please show the Modelica block where this sqrt is used?

[image]

(The when conditions are not final and will be changed)