dynawo/dynawo #3167: BaseGeneratorSynchronous saturation modeling

martinscripts, 2024-03-12:
Hey dynawo! 
I am trying to include saturation parameters into a synchronous generator model. 

The high level parameters for saturation are md, nd, mq, nq.

The parameters I have are 
[image]
according to kundur's saturation modeling
[image]
(eq. 3.189)

This is the implementation that I found deep down
```
// Mutual inductances saturation
    lambdaADPu = MdSatPPu * (idPu + ifPu + iDPu);
    lambdaAQPu = MqSatPPu * (iqPu + iQ1Pu + iQ2Pu);
    lambdaAirGapPu = sqrt(lambdaADPu ^ 2 + lambdaAQPu ^ 2);
    mdsPu = MdPPu / (1 + md * lambdaAirGapPu ^ nd);
    mqsPu = MqPPu / (1 + mq * lambdaAirGapPu ^ nq);
    cos2Eta = lambdaADPu ^ 2 / lambdaAirGapPu ^ 2;
    sin2Eta = lambdaAQPu ^ 2 / lambdaAirGapPu ^ 2;
    miPu = mdsPu * cos2Eta + mqsPu * sin2Eta;
    MdSatPPu = miPu + MsalPu * sin2Eta;
    MqSatPPu = miPu - MsalPu * cos2Eta;
```

I was assuming that 
nd = nq = B_sat
and
md = mq = A_sat

but I am not sure and this does not include the $\Psi_\mathrm{T1}$ parameter.

Does anyone know details about the implementation and how to map the parameters? 

Thank you!
.Martin

joyelfeghali, 2024-03-12:
Hello Martin!
The saturation model we are using is based on the model proposed in "Model of generator saturation for use in power-system studies" for Shackshaft and Henser. It is a bit different than the one proposed in the Kundur. 
I am looking for parameters equivalence between these two models and will get back to you.

martinscripts, 2024-04-22:
Hey @joyelfeghali, are there any news concerning saturation modeling? 
Best regards,
.Martin

joyelfeghali, 2024-04-22:
Hello @martinscripts ! 
I compared the two models and I can say that md = mq = A_sat.
But for nd and nq I still have some doubts, I have values for nq, nq between 5 and 10. There should be a coefficient between the two parameters since the equations are not exactly equivalent.
I am still analyzing both models, I will get back to you as soon as possible

martinscripts, 2024-05-28:
Hey @joyelfeghali, 

I also looked into the topic a bit and I think I don't agree that md=A_sat. 

It seems like in kundur the parameters are used to calculate $\Psi_l$, the portion of the flux linkage that gets changed because of saturation. 

In Shackshaft1979 the parameters are used to manipulate the mutual inductance, which then is used to calculate the flux linkage from the currents. 

If I missed something, please let me know. 

BR :)