dynawo/dynawo #3210: Simulation fails for the Dynawo Modelica IEC examples

joyelfeghali, 2024-05-13:
The problem is seen in the 2015 test cases of Dynawo.Examples.Wind.IEC.Neplan (#3172 ) : 
Having all switches as multiswitch makes OM simulation fails (the actual case). When replacing multiSwitch (as in 2020 model) with tripleSwitch and fiveSwitch  (simplified models of multiswitch), the OM simulation and the dynawo simulation work. 
We tried replacing equations of PQCalculus block with Modelica.ComplexBlocks.ComplexMath.Product then Modelica.ComplexBlocks.ComplexMath.ComplexToReal, this made dynawo simulation fails...

For 2020 test cases : 
the log events added to Grid protection cause an error (using v1.18 OMEdit)

rosiereflo, 2024-05-13:
Hi, I'm not sure I understand what should be investigated here?

joyelfeghali, 2024-05-13:
> Hi, I'm not sure I understand what should be investigated here?

The simulations of the Modelica examples Wind IEC 2015 that were added in #3172 are failing. The error was pointing to multiswitches but in Wind IEC 2020 this didn't cause a problem. When simplifying the multiswitches the simulation worked. But this modification was not integrated (the current version of the examples fails). 

Also simulations of Wind IEC 2020 examples are failing due since log events were added to Grid Protection (#3133 )