dynawo/dynawo #3296: error message when building models

martinscripts, 2024-06-24:
Hey dynawo team, 

I am currently integrating new models into dynawo.

After a while I found that in the .log an error say that the new model is mentioned in package.order but does not appear in package.mo. 

It turns out that the problem was not package.mo, but CMakeLists.txt.

For me, in one case the add_subdirectory() was missing and in another case the MyNewlyAddedModel.mo.

Could you adapt the error message and make it point to the CMakeLists file? That would save time when debugging.

Thanks! 
.martin

rosiereflo, 2024-07-01:
Hi, unfortunately the model compilation done by OpenModelica and the CMake files are 2 independant processes, it is thus quite difficult to catch this and add a proper error message

martinscripts, 2024-07-01:
@rosiereflo, okay, thanks. Might be worth to add it to a *frequent problems* section in the manual if there is no way to give it a meaningful error message?

rosiereflo, 2024-07-01:
yes of course! I will modify the title of this issue and we'll take care of that. Thanks for the proposal.