dynawo/dynawo #3345: Option to have logs without timestamp

gautierbureau, 2024-07-17:
We would use this option to generate logs reference for nrt. Then we would need to generate all nrt logs with this and it would be easier to compare/track changes from this point on

gautierbureau, 2024-07-17:
Adaptation in nrt script comparison would be needed

rosiereflo, 2024-07-17:
only the number of variables and the warning/Error lines are compared today as the log content is very dependent on the platform

gautierbureau, 2024-07-18:
> only the number of variables and the warning/Error lines are compared today as the log content is very dependent on the platform

I know but tools to compare text files does not handle well teh comparison. Any difference when updating the reference would allow to see by look what changed. And the timestamp does not have any interest in a reference anyway

rosiereflo, 2024-07-19:
I'm not convinced as it would require an additional action for all developers when overriding the results for a few usage of comparison that can be easily done with grep, or with vi for example you can remove the timestamp very easily for comparison. So from my current point of view the gain are less than the constraints obtained

gautierbureau, 2024-07-19:
I don't see any constraint if when running nrt the option is activated by default... The comparison would be better in github also when updating... I really don't see why you think it would require extra work for anybody

rosiereflo, 2024-07-19:
the extra work I see is at the moment of replacing the logs, which option of nrt are you referring to here?

rosiereflo, 2024-07-19:
In my understanding when replacing a reference log we would need to remove the timestamp and this is extra work for everyone. the nrt is not used to replace any reference file today, maybe you are thinking to the nrt-ref script? Unless I have misunderstood something. Instead of replying with this tone, please just explain me what was the flow you were thinking about and where I am mistaken in my understanding.