dynawo/dynawo #3419: High RAM usage during compilation of models PhotovoltaicsWeccVoltageSourceA and B

FredericSabot, 2024-10-07:


rosiereflo, 2024-10-07:
Just a small confirmation: this happens in release compilation ?

FredericSabot, 2024-10-07:
In release yes, didn't test in Debug,

rosiereflo, 2024-10-07:
ok thanks!

rosiereflo, 2024-11-05:
Hi @FredericSabot, the branch 3419_High_RAM_usage_during_compilation_of_models_PhotovoltaicsWeccVoltageSourceA_and_B seems to be solving the issue on the docker we used to run the tests. Could you please test it in your environment and let us know if the issue is solved as well? Thanks!

FredericSabot, 2024-11-06:
Hello, the branch compiles properly on my machine (with around 4Go peak ram usage with 4 cores compared to 15+ before).

rosiereflo, 2024-11-06:
> Hello, the branch compiles properly on my machine (with around 4Go peak ram usage with 4 cores compared to 15+ before).

great thanks!

closed 2024-11-15 by gautierbureau