dynawo/dynawo #3437: Problem compiling Modelica models following a modification to the initialization model

louisg-rte, 2024-10-21:
Following issue https://github.com/dynawo/dynawo/issues/3414 I've made a development which is in branch https://github.com/dynawo/dynawo/tree/3414_INIT_IEC_models . The initialisation model of the IEC models is modified and it is then impossible to compile in Dynawo 
Here is the log file that I got:
[IECWPP4ACurrentSource2015.log](https://github.com/user-attachments/files/17460252/IECWPP4ACurrentSource2015.log)

@rosiereflo

rosiereflo, 2024-11-15:
Hi, could you please try with the branch 3437_parameter_table and let me know if the results are ok? Thanks

louisg-rte, 2024-11-29:
Hi, thank you for this. The models are now compiling but all the examples fail when trying to run the models.
[image]

rosiereflo, 2024-11-29:
ok thanks let me take a look

rosiereflo, 2025-01-10:
Dyanwo do not support the algorithm keyword directly in the class. Could you please try to use a function instead?

louisg-rte, 2025-01-10:
Hi, thanks for this advice. It seems to work now !

closed 2025-01-10 by louisg-rte