dynawo/dynawo #3461: [COLIB-NGC] WT4CurrentSource_INIT current calculation

martinscripts, 2024-11-11:
Hey Joy, 

I am looking through and partly restructuring the IEC WT models, so I am able to integrate the new Type3 models without repeating code. 

I habe a question about WT3CurrentSource_INIT.mo:

initial generator side currents are calculated by 

```
IGsRe0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * cos(UPhase0) - Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * sin(UPhase0);
IGsIm0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * sin(UPhase0) + Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * cos(UPhase0);
```

P0Pu ist the grid side power, whilde IGsRe0Pu is generator side current. 
As I understand it, Re and Im components of Compelx(P,Q) are rotated to the generator system's frame and used as the current values.
I would say that the admittance currents are missing here. 
Should it be something like 

```
IGsRe0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * cos(UPhase0) - Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * sin(UPhase0) - Real(Ugs*Y);

IGsIm0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * sin(UPhase0) + Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * cos(UPhase0) - Imag(Ugs*Y);
```

?
 
Best regards,
.Martin

joyelfeghali, 2024-11-12:
Hello Martin, we are looking into that.

louisg-rte, 2024-11-14:
In my understanding, the current equations are correct.

iGsPu is the converter side current which is calculated with a rotation on the controlled current ipCmdPu and iqCmdPu (block RefFrameRotation):
`  [iGsRePu; iGsImPu] = [cos(theta), -sin(theta); sin(theta), cos(theta)] * [ipCmdPu; iqCmdPu];`

Then the grid side current is calculated based on the converter side current (block ElecSystem):
`  Complex(iGsRePu, iGsImPu) = Complex(iWtRePu, iWtImPu) + Complex(GesPu, BesPu) * Complex(uGsRePu, uGsImPu);`

which gives at initialization (WT4CurrentSource_INIT):
- ipCmdPu and iqCmdPu are defined to track the active and reactive power on grid side, so are initialized with (P0,Q0,U0) on grid side:
  `ipCmdPu(start = -P0Pu * SystemBase.SnRef / (SNom * U0Pu))`
  `iqCmdPu(start = Q0Pu * SystemBase.SnRef / (SNom * U0Pu))`

- `IGsRe0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * cos(UPhase0) - Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * sin(UPhase0);`
  `IGsIm0Pu = (-P0Pu * SystemBase.SnRef / (SNom * U0Pu)) * sin(UPhase0) + Q0Pu * SystemBase.SnRef / (SNom * U0Pu) * cos(UPhase0);`

- `Complex(IGsRe0Pu, IGsIm0Pu) = (-i0Pu * SystemBase.SnRef / SNom) + Complex(GesPu, BesPu) * Complex(UGsRe0Pu, UGsIm0Pu);`

I remain available to discuss this further

martinscripts, 2024-11-14:
Hey @louisg-rte, 
I drew the setup to clarify it for myself. 
[image]

If we use 
```
ipCmdPu(start = -P0Pu * SystemBase.SnRef / (SNom * U0Pu))
iqCmdPu(start = Q0Pu * SystemBase.SnRef / (SNom * U0Pu))
```
dont we then ignore the difference between igs and iWT? 

Rotation is done with the angle of uWt. assuming theta=0 for simplicity, we have 
ipcmd=igs,re
iqcmd=igs,im

iWT,re = -P0/(SNom*UWT) 
iWT,im = Q0/(SNom*UWT)

if we set ipcmd = iWT,re then we wouldn't have steady state. 

Does that make sense? What am I missing here?

martinscripts, 2024-11-19:
Hey @louisg-rte, 

you're welcome and also thanks for the investigations. 
I am currently implementing the IEC type 3 wind turbine model for Dynawo #3115.
Since I will soon be implementing the initialization, we could also discuss this issue at some point, if you like.

.Martin

martinscripts, 2024-11-19:
Hey @louisg-rte,

you're welcome and also thanks for the investigations.
I am currently implementing the IEC type 3 wind turbine model for Dynawo #3115 , based on the existing Type 4 model where possible. I am also doing some minor restructuring.
Since I will soon be implementing the initialization, we could also discuss this issue at some point, if you like.

.Martin

martinscripts, 2024-12-12:
Hi @louisg-rte,

I think one issue is the used angle for the reference frame rotation. 
When initializing the model with U0Pu and UPhase0, then we use the angle from the grid side, after the impedance and admittance.
This introduces an error because the voltage angles and amplitudes can differ.