dynawo/dynawo #3488: Invalid access of x_scaled protected component

casella, 2024-11-29:
Many models of the Dynawo library fail in the latest OMC because of invalid access to protected elements. This is not allowed in Modelica.

For example, [Dynawo.Examples.BESS.WECC.BESScbCurrentSource](https://libraries.openmodelica.org/branches/master/Dynawo_dev/files/Dynawo_dev_Dynawo.Examples.BESS.WECC.BESScbCurrentSource.err) fails with:
```
[Dynawo 1.7.0-master/Electrical/Controls/WECC/REPC/REPCa.mo:115:83-115:110:writable] Notification: From here:
[Modelica 3.2.3+maint.om/Blocks/Continuous.mo:1299:5-1299:47:writable] Error: Protected element 'x_scaled' may not be modified, got 'x_scaled(start = {QInj0Pu})'.
```

casella, 2024-11-29:
This is the offending code:

https://github.com/dynawo/dynawo/blob/006270995685bae8d6e29580bdc7620492b3ae29/dynawo/sources/Models/Modelica/Dynawo/Electrical/Controls/WECC/REPC/REPCa.mo#L115-L116

There are three modifiers set in the leadLag block:
```modelica
x_scaled(start = {QInj0Pu}),
x_start = {QInj0Pu},
y_start = QInj0Pu
```
as far as I understand, as the three variables are basically the same, there is no need to apply those three modifiers, which are redundant. #3490 removes that.

I see that the leadLag block initType is set to `noInit`. If the intent is to initialize the block so it has the output QInj0Pu, I would recommend to set it to initialOutput, which would further make the illegal modifier redundant.

@joyelfeghali do you agree?

ErwanGuichard49, 2024-12-02:
The modifier x_scaled(start = {QInj0Pu}) has been added for simulations under Dynawo, which ignores initial equations and thus the parameter initType.

I suggest replacing Modelica.Blocks.Continuous.TransferFunction by Dynawo.NonElectrical.Blocks.Continuous.TransferFunction in the whole opebn-source Dynawo library.

ErwanGuichard49, 2024-12-02:
I have just pushed a draft pull request [#3488](https://github.com/dynawo/dynawo/pull/3493) changing all TransferFunction blocks to the Dynawo block. The preassembled models which include REPCa.mo are not compiled, every other preassembled with the Dynawo TransferFunction is successfully compiled.

@rosiereflo could you investigate the compilation failure of (for instance) PhotovoltaicsWeccVoltageSourceB ?

rosiereflo, 2024-12-03:
on it

ErwanGuichard49, 2024-12-03:
The preassembled models which include REPCa.mo are the following ones :

BESScbWeccCurrentSource
PhotovoltaicsWeccVoltageSourceB
WTG4AWeccCurrentSource
WTG4BWeccCurrentSource

On branch **3488_TransferFunction**, none of them is compiled successfully.