dynawo/dynawo #3489: Converting Dynawo 1.7.0 to use MSL 4.0.0

casella, 2024-11-29:
The dependency of Dynawo on MSL 3.2.3 is really bad, as this library is now quite obsolete. MSL 4.0.0 was released in 2020, and MSL 4.1.0, which is fully backwards compatible, will be released at the beginning of 2025.

I tried the conversion with the automatic conversion script, this is the report of what needs to be changed:
```
Conversion removed modifier limitsAtInit in Dynawo.Electrical.Controls.WECC.REEC.REECc.
Converting class name Modelica.SIunits to Modelica.Units.SI in Modelica.SIunits.Time Dynawo.NonElectrical.Blocks.MathBoolean.OffDelay.tDelay.
Converting class name Modelica.SIunits to Modelica.Units.SI in Modelica.SIunits.Time Dynawo.NonElectrical.Blocks.MathBoolean.OffDelay.tNext.
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Sources.InjectorIDQ](modelica://Dynawo.Electrical.Sources.InjectorIDQ).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Controls.WECC.Utilities.Measurements](modelica://Dynawo.Electrical.Controls.WECC.Utilities.Measurements).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Controls.WECC.BaseControls.LineDropCompensation](modelica://Dynawo.Electrical.Controls.WECC.BaseControls.LineDropCompensation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Controls.WECC.BaseControls.LineDropCompensation](modelica://Dynawo.Electrical.Controls.WECC.BaseControls.LineDropCompensation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.BESS.WECC.BESS_INIT](modelica://Dynawo.Electrical.BESS.WECC.BESS_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.BESS.WECC.BESS_INIT](modelica://Dynawo.Electrical.BESS.WECC.BESS_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.BESS.WECC.BESS_INIT](modelica://Dynawo.Electrical.BESS.WECC.BESS_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Buses.Bus](modelica://Dynawo.Electrical.Buses.Bus).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Buses.BusWithInit](modelica://Dynawo.Electrical.Buses.BusWithInit).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Buses.BusWithInit](modelica://Dynawo.Electrical.Buses.BusWithInit).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Buses.InfiniteBusWithImpedance_INIT](modelica://Dynawo.Electrical.Buses.InfiniteBusWithImpedance_INIT).
Converting class name Modelica.Blocks.Tables.CombiTable1D to Modelica.Blocks.Tables.CombiTable1Dv in Modelica.Blocks.Tables.CombiTable1D Dynawo.Electrical.HVDC.BaseClasses.BaseDiagramPQTerminal1.tableQInj1Max.
Converting class name Modelica.Blocks.Tables.CombiTable1D to Modelica.Blocks.Tables.CombiTable1Dv in Modelica.Blocks.Tables.CombiTable1D Dynawo.Electrical.HVDC.BaseClasses.BaseDiagramPQTerminal1.tableQInj1Min.
Converting class name Modelica.Blocks.Tables.CombiTable1D to Modelica.Blocks.Tables.CombiTable1Dv in Modelica.Blocks.Tables.CombiTable1D Dynawo.Electrical.HVDC.BaseClasses.BaseDiagramPQTerminal2.tableQInj2Max.
Converting class name Modelica.Blocks.Tables.CombiTable1D to Modelica.Blocks.Tables.CombiTable1Dv in Modelica.Blocks.Tables.CombiTable1D Dynawo.Electrical.HVDC.BaseClasses.BaseDiagramPQTerminal2.tableQInj2Min.
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.HVDC.BaseClasses.BaseHvdc](modelica://Dynawo.Electrical.HVDC.BaseClasses.BaseHvdc).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP](modelica://Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP](modelica://Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP](modelica://Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcP).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcPDangling](modelica://Dynawo.Electrical.HVDC.BaseClasses.BaseHvdcPDangling).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses.BaseTransformer](modelica://Dynawo.Electrical.Transformers.BaseClasses.BaseTransformer).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses.BaseTransformer](modelica://Dynawo.Electrical.Transformers.BaseClasses.BaseTransformer).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses.BaseTransformerVariableTap](modelica://Dynawo.Electrical.Transformers.BaseClasses.BaseTransformerVariableTap).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses.BaseTransformerVariableTap](modelica://Dynawo.Electrical.Transformers.BaseClasses.BaseTransformerVariableTap).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseGeneratorTransformer_INIT](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseGeneratorTransformer_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseTransformerVariables_INIT](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseTransformerVariables_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseTransformerVariableTapCommon_INIT](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.BaseTransformerVariableTapCommon_INIT).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.IdealTransformerTapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.IdealTransformerTapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.IdealTransformerTapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.IdealTransformerTapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation](modelica://Dynawo.Electrical.Transformers.BaseClasses_INIT.TapEstimation).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.TransformerFixedRatio](modelica://Dynawo.Electrical.Transformers.TransformerFixedRatio).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.TransformerFixedRatio](modelica://Dynawo.Electrical.Transformers.TransformerFixedRatio).
Converting class name Modelica.ComplexMath.'abs' to Modelica.ComplexMath.abs in [Dynawo.Electrical.Transformers.TransformerFixedRatioAndPhase_INIT](modelica://Dynawo.Electrical.Transformers.TransformerFixedRatioAndPhase_INIT).
[... cut ...]
Converting class name Modelica.SIunits.Conversions to Modelica.Units.Conversions in [Dynawo.Examples.RVS.TestCases.TestA.TestAAlphaBetaLoadNoRestorative](modelica://Dynawo.Examples.RVS.TestCases.TestA.TestAAlphaBetaLoadNoRestorative).
Converting class name Modelica.SIunits.Conversions to Modelica.Units.Conversions in [Dynawo.Examples.RVS.TestCases.TestA](modelica://Dynawo.Examples.RVS.TestCases.TestA).
Converting class name Modelica.SIunits.Conversions to Modelica.Units.Conversions in [Dynawo.Examples.RVS.TestCases.TestA.TestAShuntNoRestorative](modelica://Dynawo.Examples.RVS.TestCases.TestA.TestAShuntNoRestorative).
```

casella, 2024-11-29:
Most of the changes are just involving SI units. Some are trivial MSL function or block name updates. 

We should check a bit more in detail the removal of limitsAtInit and deadZoneAtInit, as they may affect the semantics of the controllers. I guess there are equivalents of those options also in MSL 4.0.0, only with different ways to set them.

joyelfeghali, 2024-12-02:
> Most of the changes are just involving SI units. Some are trivial MSL function or block name updates.
>
> We should check a bit more in detail the removal of limitsAtInit and deadZoneAtInit, as they [...]

Okay I would take care of implementing that. limitsAtInit and deadZoneAtInit should not be impacting the Dynawo simulations, it concerns a few models where this modifier was added automatically. I would be removing them for the v1.17

casella, 2025-01-09:
@joyelfeghali the list of changes shown above was produced by Dymola and could be applied manually with a text editor (the order of the operations is essential!)
- replace `Modelica.SIunits.Conversions` with `Modelica.Units.Conversions` in all source files
- replace `Modelica.SIunits` with `Modelica.Units.SI` in all source files
- replace `Modelica.ComplexMath.'abs'` with `Modelica.ComplexMath.abs` in all source files
- replace `Modelica.Blocks.Tables.CombiTable1D` with `Modelica.Blocks.Tables.CombiTable1Dv` in all source files

As to the limitsAtInit and deadZoneAtInit modifiers, you can update them to something suitable as you wrote.

In principle, it should be possible to do this automatically. I've tried with Dymola and OMC, but both are messing up quite a bit with formatting and comments. I just opened OpenModelica/OpenModelica#13447, maybe we can fix OMC so it can do it without modifying the formatting at all.

casella, 2025-01-09:
As I understand, there should be no modification in the C code generated from the compilation of Dynawo models after these changes.

In case someone used Dynawo to build Modelica models of systems, they should not need to be changed to run with the new library, except that they must be updated to use MSL 4.0.0 instead of MSL 3.2.3. If the models only use Dynawo and do now use MSL explicitly, this can be done simply by editing the uses annotation. Otherwise, the conversion from MSL 3.2.3 to MSL 4.0.0 of the parts of the model that use the MSL explicitly should be performed.

casella, 2025-01-14:
OMC should now be able to do the automatic conversion without touching the formatting otherwise. I'll check that ASAP with new nightly build.

casella, 2025-01-16:
Still not good enough, the diff algorithm is now called, but it fails in 96% of the cases. I reopened https://github.com/OpenModelica/OpenModelica/issues/13447

joyelfeghali, 2025-01-16:
> Still not good enough, the diff algorithm is now called, but it fails in 96% of the cases. I reopened [OpenModelica/OpenModelica#13447](https://github.com/OpenModelica/OpenModelica/issues/13447)

Okay noted, I will be waiting to have a better handling of the automatic conversion

casella, 2025-01-20:
The failure is probably due to some issues with the parsing of C-style comments. @adrpo is looking into that, hopefully that is the only problem.