dynawo/dynawo #3517: IEC WPP q controller bug

martinscripts, 2024-12-19:
Hey @joyelfeghali, 

could it be that there is a bug in the existing Dynawo.Electrical.Controls.IEC.IEC61400.BaseControls.WPP.WPPQControl2020?

The connection should be like in the image below but actually is where I put the red mark:

[image]

I changed this in the latest work for the IEC Type 3 wind turbine model but this may be up for discussion. 

BR,
.Martin

louisg-rte, 2025-01-07:
Hello @martinscripts , I think you're totally right, this is a mistake. This should not impact our examples because we choose KwpqRef = 0

martinscripts, 2025-01-07:
Hey @louisg-rte, 
it seems like it does, unfortunately. 
Not in the master branch, strangely, but in the branch that includes the newly developed Type3 models. 
So far I can't make sense of it. If you have an idea, please share ;). 

I am using the example Neplan.WPP4ACurrentSource2020.

On this branch: 
https://github.com/martinscripts/dynawo/tree/3115_windType3

Although the gain is =0, simulation suddenly fails at 6.48 seconds. 

Can you at least reproduce this?

[image]

Warm regards and happy new year,
.Martin

louisg-rte, 2025-01-07:
Hi Martin, I didn't succeed to reproduce this error using your branch. I have no idea what could be the explanation. This may be linked to OpenModelica itself, what version are you using ? I'm using 1.18 (which is a bit old)
[image]

Happy new year,
Louis

martinscripts, 2025-01-08:
Hey @louisg-rte, 

I am also on v1.18.0.

closed 2025-01-08 by joyelfeghali