dynawo/dynawo #3614: Clarify the model extraction methodology using OMC

casella, 2025-04-01:
Currently, the generation of C code of a certain component for the Dynawo software from the Dynawo library is performed as follows:
- the component is instantiated in a model;
- equations setting to zero the derivative of all component inputs as well as voltage in terminal connectors are added; inputs are currently not marked as such in the Dynawo library, but this is irrelevant for the current discussion;
- the following compile flags are set: `--daeMode --allowNonStandardModelica=implicitParameterStartAttribute -d=disableSingleFlowEq`: the first parameter generates residual equations, the second allows to handle parameters without a default value nor a start attribute, the third inhibits the front-end from creating default connection equations that would set the terminal connector currents to zero;
- the model is compiled;
- the code for the computation of residuals is fetched from the `modelname_16dae.c` C source file, removing the unnecessary equations setting the derivatives to zero.

This is demonstrated by the following example:
```modelica
model LoadAlphaBetaRestorative
  Dynawo.Electrical.Loads.LoadAlphaBetaRestorative load;
equation
der(load.terminal.V.im) =0;
der(load.terminal.V.re) =0;
der(load.PRefPu) =0;
der(load.QRefPu) =0;
der(load.deltaP) =0;
der(load.deltaQ) =0;
 when(time > 999999) then
    load.switchOffSignal1.value = false;
    load.switchOffSignal2.value = false;
 end when;
annotation(
    uses(Dynawo(version="1.8.0")),
    experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.002),
    __OpenModelica_commandLineOptions = "--daeMode -d=disableSingleFlowEq --allowNonStandardModelica=implicitParameterStartAttribute");
end LoadAlphaBetaRestorative;
```
The generated code should compute the equation residuals including those input and voltage terminal variables, without explicitly assigning them any specific value. The above-described methods achieves this goal because:
1. the added `der(x) = 0` equations are exactly the number that is required to make the system square, i.e. the number of inputs plus the number of effort (or flow, they are the same) variables in the terminal connector
2. writing der(x) makes x a state variable, which means that x is assumed to be known when generating the residual equations, but is not assigned any value, because the value of x in any expression will be fetched from the state vector

Point 2. is exactly what is needed to handle component inputs. Regarding the terminal connector variables, it sets the voltage as known, so when the residual equations are generated, they may unnecessarily be re-written in a form that computes the current given the voltage, which is not good for Dynawo. This is demonstrated by the following MWE:

```modelica
package TestExtractionDAE
  connector Pin
    Real V;
    flow Real i;
  end Pin;
  
  connector InputReal = input Real;
  
  model VariableResistanceLoad
    Pin p;
    parameter Real P;
    InputReal deltaP;
    Real V;
    Real i;
  equation
    V = p.V;
    i = p.i;
    P + deltaP = V*i;
  end VariableResistanceLoad;
  
  model VariableResistanceLoadExtractor
    VariableResistanceLoad load;
  equation
    der(load.p.V) = 0;
    der(load.deltaP) = 0;
  annotation(__OpenModelica_commandLineOptions = "--daeMode --allowNonStandardModelica=implicitParameterStartAttribute -d=disableSingleFlowEq",
  experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.002));
  end VariableResistanceLoadExtractor;
end TestExtractionDAE;
```
The debugger shows:

[image]

which is not what would be required for Dynawo: the power flow equation is explicitly solved for the current given the voltage, so it is not part of the set of residual equations, which is passed to the IDA solver, but is rather part of the other, explicitly solved equations that are only computed once, after the IDA step is finalized.

casella, 2025-04-01:
What you should do is to slightly change point 2. of the procedure sketched above as follows:

2a. equations setting to zero the derivative of all component inputs are added;
2b. equations with a dummy function relating the voltage and currents of each connector are added.

2.b corresponds to connecting every terminal to a generic one-port component, that enforces a generic voltage-current relationship, as I mentioned during today's meeting. Later one you will of course remove both the `der(x) = 0` and `dummy(v,i) = 0` residuals from the C code, and only keep the actual component residuals.

This is demonstrated by the following MWE:
```modelica
package TestExtractionDAENew
  connector Pin
    Real V;
    flow Real i;
  end Pin;
  
  connector InputReal = input Real;
  
  model VariableResistanceLoad
    Pin p;
    parameter Real P;
    InputReal deltaP;
    Real V;
    Real i;
  equation
    V = p.V;
    i = p.i;
    P + deltaP = V*i;
  end VariableResistanceLoad;
  
  function dummy
    input Real V;
    input Real i;
    output Real y;
  algorithm
    y := V*i; // The actual algorithm doesn't matter
  annotation(Evaluate = false);
  end dummy;
  
  model VariableResistanceLoadExtractor
    VariableResistanceLoad load;
  equation
    der(load.deltaP) = 0;
    dummy(load.p.V, load.p.i) = 0;
  annotation(__OpenModelica_commandLineOptions = "--daeMode --allowNonStandardModelica=implicitParameterStartAttribute -d=disableSingleFlowEq",
  experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.002));
  end VariableResistanceLoadExtractor;
end TestExtractionDAENew;
```
Note that the actual algorithm of function dummy is totally irrelevant: what is important is that the function is _not_ inlined, so it cannot be solved explicitly and it will enforce a generic relationship between connector voltage and connector current in the adjacency matrix. Coupled with the equation relating voltage and current inside the component, this will create a strong component (implicit system of algebraic equations) in the extractor model that the backend will not be able to solve explicitly, so it will actually lead to the generation of residual equations to be passed on to the IDA solver:

[image]

Of course then you will discard the equation with the dummy function call, and only keep the one from the component.

For the Dynawo connectors, you will have to use the following dummy function:

```modelica
function dummy
  input Complex V;
  input Complex i;
  output Complex y;
algorithm
  y := V*i; // The actual algorithm doesn't matter
annotation(Evaluate = false);
end dummy;
```
and define the extractor model as follows:
```modelica
model LoadAlphaBetaRestorative
  Dynawo.Electrical.Loads.LoadAlphaBetaRestorative load;
equation
dummy(load.terminal.V, load.terminal.i) = Complex(0);
der(load.PRefPu) =0;
der(load.QRefPu) =0;
der(load.deltaP) =0;
der(load.deltaQ) =0;
 when(time > 999999) then
    load.switchOffSignal1.value = false;
    load.switchOffSignal2.value = false;
 end when;
annotation(
    uses(Dynawo(version="1.8.0")),
    experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.002),
    __OpenModelica_commandLineOptions = "--daeMode -d=disableSingleFlowEq --allowNonStandardModelica=implicitParameterStartAttribute");
end LoadAlphaBetaRestorative;
```
This will ensure that you always get the required C code also with future versions of OMC 😃

casella, 2025-04-01:
Adding @rosiereflo, @marcochiaramello, @joyelfeghali, @phannebohm, @bernardbachmann to the loop.

rosiereflo, 2025-04-02:
Thanks for this proposal! I will try it on the branch using the v1.24.5 version of OpenModelica and let you know the results

marcochiaramello, 2025-04-02:
It is an interesting proposal indeed, we'll try this out!

casella, 2025-04-02:
I haven't checked what happens with Jacobian, but I assume everything's going to be fine.

It could also be that using der(terminal.v) = 0 also happens to work, as long as terminal.v is a Complex record and not a scalar variables, so the backend is not aggressive enough to try to solve for the Complex current explicitly. But that is dangerous, and may stop working in the future. Adding the dummy constraint seems to me a safer option, from a structural analysis point of view.

phannebohm, 2025-04-02:
> This will ensure that you always get the required C code also with future versions of OMC 😃

That is until omc simplifies states `x` with `der(x) = 0` to be constants and replaces them with their start values... 😅

I wonder if you could also use a dummy function for this instead. Something like this
```modelica
function dummy1
  input Real x;
  output Real y;
algorithm
  y := x + x; // The actual algorithm doesn't matter
annotation(Evaluate = false);
end dummy1;
```
and then writing
```modelica
  0 = dummy1(load.PRefPu);
```
will also force the compiler to assume `load.PRefPu` is solved in this equation and known in the other equations.

Or is it important that those variables are interpreted as states by the compiler?

> I haven't checked what happens with Jacobian, but I assume everything's going to be fine.

For the Jacobian omc in fact seems to look into the algorithm of `dummy()` and inline the derivative of that, but it doesn't seem to interfere with the rest of the system from what I can see in my quick interrogation of `TestExtractionDAENew`.

casella, 2025-04-02:
> > This will ensure that you always get the required C code also with future versions of OMC 😃
> 
> That is until omc simplifies states `x` with `der(x) = 0` to be constants and replaces them with their start values... 😅

Yeah. And possibly replace der(x) = k with x = x.start with k*time. Are there plans to do that? 😃 

> I wonder if you could also use a dummy function for this instead. Something like this
> 
> function dummy1
>   input Real x;
>   output Real y;
> algorithm
>   y := x + x; // The actual algorithm doesn't matter
> annotation(Evaluate = false);
> end dummy1;
> 
> and then writing
> 
>   0 = dummy1(load.PRefPu);
> 
> will also force the compiler to assume `load.PRefPu` is solved in this equation and known in the other equations.

Indeed. Sounds like a good idea.

> Or is it important that those variables are interpreted as states by the compiler?

I don't think so. In fact, that could have been a weak point of the old approach.

> For the Jacobian omc in fact seems to look into the algorithm of `dummy()` and inline the derivative of that, but it doesn't seem to interfere with the rest of the system from what I can see in my [...]

It goes without saying that the derivatives of the dummy equations should also be discarded.

phannebohm, 2025-04-03:
> Yeah. And possibly replace `der(x) = k` with `x = x.start + k*(time - StartTime)`. Are there plans to do that? 😃

Right, simple things like that. There are no plans for the near future. But it depends on how many students want to do a bachelor thesis with us I guess...

> It goes without saying that the derivatives of the dummy equations should also be discarded.

Sure, but we need to make sure the compiler doesn't play smart and substitute into other equations, just to be super paranoid and safe. One quick (and incomplete) way to check this is to change the algorithm inside the dummy function and see if anything else changes in the generated code.

casella, 2025-04-03:
> > Yeah. And possibly replace `der(x) = k` with `x = x.start + k*(time - StartTime)`. Are there plans to do that? 😃
> 
> Right, simple things like that. There are no plans for the near future. But it depends on how many students want to do a bachelor thesis with us I guess...

😄 

Seriously, solving _differential_ equations simbolically is a big deal, I am not aware of any Modelica tool doing that. It's something that we should probably discuss within MAP-Lang first, although I think the specification doesn't say anything about it.

> Sure, but we need to make sure the compiler doesn't play smart and substitute into other equations, just to be super paranoid and safe. One quick (and incomplete) way to check this is to change the [...]

Yep. But I guess, as long as you forbid inlining, there's not much playing smart left.

One way could be to put failing assertions in the dummy function algorithms, so the generated code would fail immediately if there is any trace of them left in the selected C code. What do you think?

casella, 2025-09-25:
Updated MWE:
```modelica
package TestExtractionDAECorrect
  connector Pin
    Real V;
    flow Real i;
  end Pin;
  
  connector InputReal = input Real;
  
  model VariableResistanceLoad
    Pin p;
    parameter Real P;
    InputReal deltaP;
    Real V;
    Real i;
  equation
    V = p.V;
    i = p.i;
    P + deltaP = V*i;
  end VariableResistanceLoad;
  
  function dummy1
    input Real u1;
    input Real u2;
    input Real u3;
    output Real y;
  algorithm // doesn't really matter, the function call will eventually be removed from the generated code
    y := u1 + u2 + u3;
  annotation(Evaluate = false);
  end dummy1;

  function dummy2
    input Real u1;
    input Real u2;
    input Real u3;
    output Real y;
  algorithm // doesn't really matter, the function call will eventually be removed from the generated code
    y := u1 + u2 + u3;
  annotation(Evaluate = false);
  end dummy2;
  
  model VariableResistanceLoadExtractor
    VariableResistanceLoad load;
  equation
    0 = dummy1(load.deltaP, load.p.V, load.p.i);  // generated for the load.deltaP input connector
    0 = dummy2(load.deltaP, load.p.V, load.p.i);  // generated for the flow variable load.p.i of the connector load.p
  annotation(__OpenModelica_commandLineOptions = "--daeMode --allowNonStandardModelica=implicitParameterStartAttribute -d=disableSingleFlowEq --generateDynamicJacobian=symbolic",
  experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.002));
  end VariableResistanceLoadExtractor;
end TestExtractionDAECorrect;
```
The 16dae.c file contains:

```
$DAEres0 = TestExtractionDAECorrect.dummy1(load.deltaP, load.V, load.i)
$DAEres1 = TestExtractionDAECorrect.dummy2(load.deltaP, load.V, load.i)
$DAEres2 = load.deltaP + load.P - load.V * load.i
```
The first two are dummy equations and must be removed. What remains is the actual interesting equation. The 12jac.c file contains the corresponding
```
$DAEres2.$pDERA.dummyVarA = load.deltaP.SeedA + (-load.V) * load.i.SeedA - load.V.SeedA * load.i
```