dynawo/dynawo #3649: Python event handler fails on branch 3301_rt_dynawo_experiments

lueder-kws, 2025-05-05:
Following steps 1 to 4 of _example_simulationRT/example.md_ python event handler fails in step 4:
```bash
lueder@simt-dynawo22:~/github/dynawo/example_simulationRT$ python3 -i python_launcher/src/main_interactive.py -d RTNordic_example/Events.dyd -p RTNordic_example/Events.par
Traceback (most recent call last):
  File "/home/lueder/github/dynawo/example_simulationRT/python_launcher/src/main_interactive.py", line 38, in <module>
    events = load_events(dyd_file, par_file)
  File "/home/lueder/github/dynawo/example_simulationRT/python_launcher/src/main_interactive.py", line 14, in load_events
    dyd_handler = DydHandler(dyd_file)
  File "/home/lueder/github/dynawo/example_simulationRT/python_launcher/src/handlers/xml_handler.py", line 20, in __init__
    schema_root = etree.parse(self.XSD_FILE)
  File "src/lxml/etree.pyx", line 3536, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1876, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1902, in lxml.etree._parseDocumentFromURL
  File "src/lxml/parser.pxi", line 1805, in lxml.etree._parseDocFromFile
  File "src/lxml/parser.pxi", line 1177, in lxml.etree._BaseParser._parseDocFromFile
  File "src/lxml/parser.pxi", line 615, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 725, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 652, in lxml.etree._raiseParseError
OSError: Error reading file '/home/vermeulenthi/Projects/dynawo/dynawo/dynawo/sources/API/DYD/xsd/dyd.xsd': failed to load external entity "/home/vermeulenthi/Projects/dynawo/dynawo/dynawo/sources/API/DYD/xsd/dyd.xsd"
>>>
```
Probably, some some changes in _example_simulationRT/python_launcher/src/utils.py_ will fix the problem.

lueder-kws, 2025-05-06:
Sorry, after pull there were no more problems building branch 3301. But example_simulationRT/example.md disappeared. 
Where do I find some documentation how to operate dynawo in Real Time mode ?

vermeulenthi, 2025-05-12:
Hi @lueder-kws, sorry for the delay.
The example is back in the 3301 branch, with a little more documentation.
Do not hesitate to contact me for questions or feedback on this real-time mode.

lueder-kws, 2025-06-05:
Thanks for the detailed documentation of Dynawo RT.
So it is time to close this issue

closed 2025-06-05 by lueder-kws